hzeller/beagleg #6: Kernel 4.1+ remoteproc

leonardoromor, 2016-03-15:
Hello @hzeller , 
I was wondering if you are working on a version that supports the new kernel feature remoteproc to communicate with the PRUs and if not, if you would like to have that feature.

hzeller, 2016-03-15:
Given that this is a fairly new feature, it is probably dangerous to rely on as we then have to maintain two versions of the code to support older kernels. In particular as I can't see a particular advantage to just using the TI PRU library that does whatever is needed internally to start/stop the PRU.

The communication with the PRU is very simple and straightforward, so using a kernel feature doesn't seem like a huge benefit unless this code is ported to other platforms; but then again, the the PRU code for whatever other embedded processor would need to be different as well.

Can you give me some insight what the benefits are from your point of view ?

leonardoromor, 2016-03-15:
> Given  that this is a fairly new feature, it is probably dangerous to rely on as we then have to maintain two versions of the code to support older kernels. 

Well from what I understood (if I understood correctly) they started working on it since 2011
http://lxr.free-electrons.com/source/drivers/remoteproc/remoteproc_core.c
and it will be the new standard, that's the only reason why I proposed a source code update
> In particular as I can't see a particular advantage to just using the TI PRU library that does whatever is needed internally to start/stop the PRU.

https://github.com/machinekit/machinekit/issues/839
https://groups.google.com/forum/#!topic/beagleboard/cyM3f935wMA

Sorry for bothering you, it's an awesome project and I really appreciate your work, maybe it's still to early to think about remoteproc.

> Can you give me some insight what the benefits are from your point of view ?

I don't have direct benefits if not that you can use newer kernels

hzeller, 2016-03-15:
Ah, I see, so it actually wouldn't work with later kernels. Agreed, in that case we'd need to support the old and new interface.

Would you like to prepare a pull request ?

leonardoromor, 2016-03-15:
Meh, I would love, when I will finish my project I will work on that to thank you. I will try to not disappoint you :D

bigguiness, 2016-09-21:
Is this still an issue? It appears the latest jessie images still support the uio_pruss driver. Is it really going to be replaced be a remoteproc driver?

hzeller, 2016-09-21:
The last kernel coming from TI I tested only supported the remoteproc thing, but it also logs into dmesg that it is not really ready. Given that the `-bone` kernels actually work with the PRUSS, we're good for now.

I think we can close this for now.

closed 2016-09-21 by hzeller