hzeller/beagleg #13: Alignment trap: not handling instruction f843ec10 at [<00014c9a>]

mikikg, 2016-06-30:
Hi,

I have tried latest BeagleG code on xenomai-r79 (machinekit) but have some problems.
It is clear GIT checkout with all defaults. 

If I run `./machine-control -p 3000 --nohoming-required -c sample.config` after several G-code commands something goes bad, got following in console:

```
INFO  [2016-06-30 17:21:06.049176] Accepting new connection from 192.168.0.230
Bus error
```

G-codes transaction looks like this in terminal:

```
telnet beaglebone-02 3000
Trying 192.168.0.6...
Connected to beaglebone-02.
Escape character is '^]'.
G01 F10 X100 M114 M114
ok
X:0.000 Y:0.000 Z:0.000 E:0.000 [ABS. MACHINE CUBE X:0.000 Y:0.000 Z:0.000] (Unsure: machine never homed!)
ok
X:0.000 Y:0.000 Z:0.000 E:0.000 [ABS. MACHINE CUBE X:0.000 Y:0.000 Z:0.000] (Unsure: machine never homed!)
ok
M114
X:100.003 Y:0.000 Z:0.000 E:0.000 [ABS. MACHINE CUBE X:100.003 Y:0.000 Z:0.000] (Unsure: machine never homed!)
ok
G01 X1 M114 M114
ok
X:100.003 Y:0.000 Z:0.000 E:0.000 [ABS. MACHINE CUBE X:100.003 Y:0.000 Z:0.000] (Unsure: machine never homed!)
ok
X:100.003 Y:0.000 Z:0.000 E:0.000 [ABS. MACHINE CUBE X:100.003 Y:0.000 Z:0.000] (Unsure: machine never homed!)
ok
Connection closed by foreign host.
```

In syslog I only get this:

```
Jun 30 17:21:51 beaglebone-02 kernel: [ 1643.269886] Alignment trap: not handling instruction f843ec10 at [<00014c9a>]
Jun 30 17:21:51 beaglebone-02 kernel: [ 1643.277375] Unhandled fault: alignment exception (0x801) at 0xb6c88036
```

Any clue what can be wrong here?

hzeller, 2016-06-30:
are you using the latest 4.4 Linux kernel by any chance?

mikikg, 2016-06-30:
No, 3.8.13

```
uname -a
3.8.13-xenomai-r79 #1 Wed Jun 15 22:12:09 UTC 2016 armv7l GNU/Linux
```

hzeller, 2016-06-30:
There was some alignment problem that showed up in newer kernels. Leonardo has a first patch for this, see https://github.com/hzeller/beagleg/pull/12  - can you try if that works for you ?
(it will not be the final solution, but worth a try)

mikikg, 2016-06-30:
With leonardo's patch seams to work, no more Bus error.

commit 633a1316 (2016-07-04): o Fix alignment issue: the compiler seems to be too clever
  when copying unaligned structs between different memory kinds.
  (Fixes #13)
o Output errno when it was not possible to open PRU.
o Remove superfluous logging to stderr.
closed 2016-07-04 by hzeller