hzeller/beagleg #16: Is there a plan to add more kinematics transformation to fit more different machine?

Dark-Guan, 2016-07-29:
Is there a plan to add more kinematics transformation to fit more different machine?

hzeller, 2016-07-29:
At this point, all developers have euclidian machines, so this is what they are focusing on. Current focus is to get the planner a little more smart, and then there are a couple of other things on the list first before worrying about different geometries.

The current implementation assumes that a line-segment in euklidian space can be decomposed into linear axis movements on all motors. Part of the fact that we can do very fast steps with almost no Host-CPU is because these linear calculations can be done very efficiently in the PRU.

Any linear geometry that can be transformed with an affine transformation should be working out of the box.

I suspect that the geometry does not work out linearly in e.g. Rostock machines, but I have never looked into that math.

Anyway, if you would like to see that in BeagleG because you have access to such machine I guess is to get involved and send patches.

Dark-Guan, 2016-08-02:
Thanks for your reply,hzeller!
I haven't read the PRU part of beagleG. So I should work hard on it,later.
Actully I am focusing on LinuxCNC and Redeem (two CNC system that can run on beaglebone black),too
And them all support different kinematics ,such delta .

I am seeking to add this feature to BeagleG.
Yes,Delta and Scara mean nonlinearity,but they can be transformed into linearity. 
So I will work on finding the possiblity to add the feature to bealgG.

Actually beagleG is efficent!