hzeller/beagleg #20: Problem about endstop setting for A B or C axis

Dark-Guan, 2016-09-07:
Actully,The endstop can be setted for A b or C axis , The motor will stop when the endstop is triggered ,but when G28 is runing ,the triggered endstop of A B or C will block the hole progress, untill I release the endstop by hand.
I think that means there is no walking-back move when running G28.
I want to know  how to fix it.

hzeller, 2016-09-07:
So the code actually should move back for any axis, essentially going back until the switch is un-triggered again (but I have never tested this on a real A, B, C axis, so maybe something is off).

Maybe it is going too slow ?

What are your settings for the A/B/C axis in your configuration file, and what are endstop switch settings there ?

bigguiness, 2016-09-07:
Also, are the endstops properly mapped?

What is the DEBUG output when you start machine-control?

Dark-Guan, 2017-02-16:
Sorry for apply so late.
I am not working on this problem , I think I should read all the code to solve this problem.

hzeller, 2022-08-03:
From the code, things should be working if the switches are properly mapped as pointed out by @bigguiness 
(start looking at `move_to_endstop()`).
My suspicion is that the homing speed for the axes might not be properly chosen for rotational axes, and should possibly be made configurable ... for which there is already a TODO:

https://github.com/hzeller/beagleg/blob/4693e81c0271b76292bcb8455824e01ee0f7ba50/src/gcode-machine-control.cc#L927-L928

Not sure if this is still a problem right now; since there was no activity on this issue for a while I'd lean towards closing it.