hzeller/beagleg #23: homing does not work when step is too large

aboitor, 2017-03-05:
I had a set-up where the step size was .053125mm

When I would run G28, homing would seem to work until it reached the endstop, but then the program would hang and there was not output to the steppers anymore. After I reduced the step size by adding more microstepping, the problem went away.

What I think was happening was that it would hit the endstop and then try to backoff with the smaller step size. Maybe due to rounding it would not actually issues any steps. I have not tried any modifications to the code, but I looked at line 698 in gcode-machine-control.cc to get an idea of what might be happening
`
  const float kHomingMM = (backoff) ? 0.5 : 0.05; // TODO: make configurable?
  const float kBackoffMM = kHomingMM / 10.0;      // TODO: make configurable?
`

hzeller, 2017-03-05:
It sounds plausible that it was actually rounded to less than one step, so no steps were emitted (and therefore hang, as it never got out of that again).

lromor, 2018-01-03:
@hzeller is this still open? If not, I can try to work on it

bigguiness, 2018-03-27:
On my hardware the steps-per-mm are 503.937 or 0.00198 mm/step. With that resolution the 0.5mm moves toward the endstop and 0.05mm moves away work great and the homing is very repeatable. The 0.05mm on my machine is approx 25 motor steps.

On hardware with courser resolution, like aboitor's 0.053125 mm/step, those fixed move steps will of course not work because the motor step size is to close to the backoff step size.

I think we need to either:

1) Make the homing and backoff step size configurable. This could get complicated on some setups and make the homing very unrepeatable.

or

2) Use the 0.5/0.05 as a starting point and then check the smaller value against the largest actual motor step size. If the motor step size is less than something like 2-3 times the backoff distance (i.e. 2 or 3 steps), the homing and backoff distances should be increased to ensure that some steps are actually emitted.

Note:
I originally used the 0.5 mm homing step size because of the distance that the buttons on my homing switches can be depressed. With that step size I can be assured that the button will always be detected before it is fully pressed and the machine crashes into the hard stops. The 1/10 smaller backoff allows the axis to move until it is just off the limit.