hzeller/beagleg #25: Explore if eHRPWM can also be used for PWM generation

hzeller, 2017-03-24:
Currently, we only support the Timer pins for PWM generation. But there is also other PWM dedicated hardware in the am335x (eHRPWM), which we also should support if possible. This would also expand the  available Pins people can use for PWM (see limit mentioned in #24)

bigguiness, 2017-04-09:
Take a look at this:
https://github.com/beagleboard/bb.org-overlays/blob/master/examples/cape-unversal-pwm.txt
Maybe we could use this approach as a start for the ehrpwm pins?

darenschwenke, 2017-04-10:
Might I suggest taking a look at:
https://github.com/machinekit/machinekit/tree/master/src/hal/drivers/hal_pru_generic

dgouramanis, 2017-04-10:
I can not get ehrpwm working at all with certain pins using Debian 8 and 4x kernel. I posted about the issue here. https://github.com/rust-embedded/rust-sysfs-pwm/issues/2