hzeller/beagleg #28: prussdrv_open fails with latest BB images

Stewels, 2017-11-23:
The program compiled from the git repo fails with the message: 

 prussdrv_open() failed (-1) No such file or directory
machine-control: pru-motion-queue.cc:175: PRUMotionQueue::PRUMotionQueue(HardwareMapping*, PruHardwareInterface*): Assertion `success' failed.
Aborted

There seems to be previous discussions and patches for PRU issues,  but nothing recent. I've followed all the instructions to the letter, including setting the overlay and starting the uio_pruss kernel module, but  nothing works, 

The only thing I've not tried is updating the kernel, since there doesn't seem the be any recent bone kernels released on the linked eLinux page. I'm currently using the kernel supplied with the latest image, namely 4.4.91-ti-r133.

hzeller, 2017-11-23:
Have a look at the Readme how to get it working. The TI kernels have by default the new, beta way of talking with the PRU installed, but that is not yet supported by BeagleG, because it is way more complicated with no gain.

Stewels, 2017-11-23:
Thanks, but you will notice on th eLinux page that there has not been a --bone kernel release for a long time (3.8.x seems to be the last one). The problem is that I need to run other software on the Beaglebone that needs a 4.4.x or later kernel... But if it is much more complicated to support the new PRU API, then I will try to find another way around it. 

Just one more comment: The test you propose in the Readme (if lsmod | grep "uio_pruss" &> /dev/null ; then echo "The kernel is BeagleG ready"; fi) comes back positive with the latest kernels, even though the PRU cannot be opened by BeagleG. Perhaps just this test can be updated then?

darenschwenke, 2017-11-23:
uio_pruss does work on the 4.4 kernels now.  I've been using it for machinekit with a BBGW to support the wifi chip.  This may just be a path issue as they changed.

lromor, 2018-02-02:
Hello @Stewels, did you managed to solve this issue?

Stewels, 2018-02-02:
> Reply-to: hzeller/beagleg <reply+02057edfd4d4807addc9383d7a34f0e16a9f4e
> 357f6922c792cf00000001168c344e92a169ce10771ef9@reply.github.com>
> images (#28)

Hi Leonardo

I tried a few simple solutions based on the suggestions in this issue,
but could not get it to work. In the end I ran out of time and rather
went to the Raspberry Pi platform using the PiGPIO library.

lromor, 2018-02-02:
Ok, there were two things that would have gone against a normal usage of BeagleG. 
One of them will be solved on the next debian release.
It has to do with the pwm generation. You will not be able to use the pwm because the timers driver is not enabled. So if you try to run beagleg, it will freeze your system.

I wrote a small patch so that you can still use it, even without pwm.

Regarding the uio_pruss module, could you please paste here what's the lsmod output?
thank you,

-l

Oleminator11, 2019-12-17:
Hello @Stewels,

I had the same error but on BBB image 3.8.x.

I found a workaround for this error, but it lasts only until you reboot the system.

`echo BB-BONE-PRU-01 > /sys/devices/bone_capemgr.8/slots`.

Here is the source: "[Equal Error](https://github.com/beagleboard/am335x_pru_package/issues/1)".

Hopefully it will help you as it helped me.

hzeller, 2022-08-03:
I've updated the documentation in the INSTALL file to explain how to make sure the PRU set-up works with the latest images.
https://github.com/hzeller/beagleg/blob/main/INSTALL.md#enable-pru
(as well as other set-ups needed to get the pins working)

closed 2022-08-03 by hzeller

deandzines, 2024-03-25:
I am getting the same error running the latest BBB image with kernel 5.10. 

Should I downgrade to an earlier kernel? If so what is the latest tested version?

**machine-control Error:**
```
ERROR [2024-02-12 23:19:51.172829] prussdrv_open() failed (-1) No such file or directory
machine-control: pru-motion-queue.cc:187: PRUMotionQueue::PRUMotionQueue(HardwareMapping*, PruHardwareInterface*): Assertion `success' failed.
Aborted
```

**uname -a:**
`Linux BeagleBone 5.10.168-ti-r71 #1bullseye SMP PREEMPT Fri Sep 1 04:05:07 UTC 2023 armv7l GNU/Linux`

**beagle-version:**
```
eeprom:[A335BNLT00C02119BBBK0FD8]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Bullseye IoT Image 2023-09-02]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2022.04-ge0d31da5 (Aug 04 2023 - 18:48:26 +0000)]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-ADC-00A0.kernel]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.kernel]
UBOOT: Loaded Overlay:[PCX-BB-01A-LED-00A0]
kernel:[5.10.168-ti-r71]
nodejs:[v12.22.12]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=PCX-BB-01A-LED.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~bullseye+20210821]
pkg:[bb-customizations]:[1.20240119.0-0~bullseye+20240119]
pkg:[bb-usb-gadgets]:[1.20231003.1-0~bullseye+20231003]
pkg:[bb-wl18xx-firmware]:[1.20230414.0-0~bullseye+20230414]
pkg:[kmod]:[28-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20210315-3]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input render bluetooth netdev i2c gpio admin tisdk weston-launch cloud9ide]
cmdline:[console=ttyS0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    9.739303] remoteproc remoteproc0: wkup_m3 is available
[   42.639162] remoteproc remoteproc0: powering up wkup_m3
[   42.639200] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   42.639491] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    0.178669] platform 4a300000.pruss: Cannot lookup hwmod 'pruss'
[    9.933356] sysfs: cannot create duplicate filename '/bus/platform/devices/4a300000.pruss'
[   53.603846] pruss_uio 4a300000.pruss: _od_fail_runtime_resume: FIXME: missing hwmod/omap_dev info
[   53.603878] pruss_uio 4a300000.pruss: pm_runtime_get_sync() failed
dmesg | grep pinctrl-single
[    9.239976] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    9.241527] gpio-of-helper ocp:cape-universal: Failed to get gpio property of 'P8_03'
[    9.241556] gpio-of-helper ocp:cape-universal: Failed to create gpio entry
[    9.936853] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0 name='P8_03'
[    9.937074] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1 name='P8_04'
[    9.937266] gpio-of-helper ocp:cape-universal: Allocated GPIO id=2 name='P8_05'
[    9.937431] gpio-of-helper ocp:cape-universal: Allocated GPIO id=3 name='P8_06'
[    9.937905] gpio-of-helper ocp:cape-universal: Allocated GPIO id=4 name='P8_07'
[    9.938100] gpio-of-helper ocp:cape-universal: Allocated GPIO id=5 name='P8_08'
[    9.938290] gpio-of-helper ocp:cape-universal: Allocated GPIO id=6 name='P8_09'
[    9.938462] gpio-of-helper ocp:cape-universal: Allocated GPIO id=7 name='P8_10'
[    9.938625] gpio-of-helper ocp:cape-universal: Allocated GPIO id=8 name='P8_11'
[    9.938903] gpio-of-helper ocp:cape-universal: Allocated GPIO id=9 name='P8_12'
[... 42 lines cut ...]
[    9.946518] gpio-of-helper ocp:cape-universal: Allocated GPIO id=52 name='P9_19'
[    9.946689] gpio-of-helper ocp:cape-universal: Allocated GPIO id=53 name='P9_20'
[    9.946978] gpio-of-helper ocp:cape-universal: Allocated GPIO id=54 name='P9_21'
[    9.947159] gpio-of-helper ocp:cape-universal: Allocated GPIO id=55 name='P9_22'
[    9.947336] gpio-of-helper ocp:cape-universal: Allocated GPIO id=56 name='P9_23'
[    9.947509] gpio-of-helper ocp:cape-universal: Allocated GPIO id=57 name='P9_24'
[    9.947697] gpio-of-helper ocp:cape-universal: Allocated GPIO id=58 name='P9_25'
[    9.947856] gpio-of-helper ocp:cape-universal: Allocated GPIO id=59 name='P9_26'
[    9.948036] gpio-of-helper ocp:cape-universal: Allocated GPIO id=60 name='P9_27'
[    9.948217] gpio-of-helper ocp:cape-universal: Allocated GPIO id=61 name='P9_28'
[    9.948400] gpio-of-helper ocp:cape-universal: Allocated GPIO id=62 name='P9_29'
[    9.948571] gpio-of-helper ocp:cape-universal: Allocated GPIO id=63 name='P9_30'
[    9.948744] gpio-of-helper ocp:cape-universal: Allocated GPIO id=64 name='P9_31'
[    9.948923] gpio-of-helper ocp:cape-universal: Allocated GPIO id=65 name='P9_41'
[    9.949092] gpio-of-helper ocp:cape-universal: Allocated GPIO id=66 name='P9_91'
[    9.949250] gpio-of-helper ocp:cape-universal: Allocated GPIO id=67 name='P9_42'
[    9.949423] gpio-of-helper ocp:cape-universal: Allocated GPIO id=68 name='P9_92'
[    9.949434] gpio-of-helper ocp:cape-universal: ready
dmesg | grep wlcore
lsusb
Bus 001 Device 003: ID 05ac:024f Apple, Inc. Aluminium Keyboard (ANSI)
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```