hzeller/beagleg #65: ESTOP should clear any internal queue

lromor, 2023-04-07:
After estop is triggered, the machine state is set to homing_state_ = GCodeMachineControl::HomingState::NEVER_HOMED;
and motors are shutdown. Segments could still be enqueued in the planner or in worse in the PRU. 

We should enforce a clear state after that or an estop clear could lead to machine executing an unfinished PRU segment (at some undefined speed).

bigguiness, 2026-01-15:
This is old but I think still relevant.

I agree, if and ESTOP occurs all queued motion needs to be flushed.

The ESTOP will (should?) disable all the axes. At that point the machine position is unknown. Any queued motion should be flushed to avoid any crash issues.

lromor, 2026-01-15:
Hi Hartley!

Yes, I feel bad I didn't find the time to solve this issue. I did try to think about a solution in the past months.
#68  should do the PRU flushing and reset. The problem right now is that once we hit the [50ms idle time](https://github.com/hzeller/beagleg/blob/main/src/common/fd-mux.h#L32),  we call [Timeout()](https://github.com/hzeller/beagleg/blob/f52f9e0c0d60d66fc1c914a0898564f6538a48d1/src/gcode-parser/gcode-streamer.cc#L42) which then calls `planner_->BringPathToHalt();` which is a blocking call that tries to gracefully consume all the remaining planned steps. This is not nice if the system receives an ESTOP because it's not handled until the queue is flushed.

A possible solution would be to avoid bringing the path to halt, if we have more than x seconds of planned motion. The clean solution would be to move to a non blocking call which ensures other events can be handled while we are flushing the queue.