hzeller/beagleg #72: Setting up a CNC Viewer and an Attempt to Parse the .service File

silver2row, 2025-01-30:
Hello Again,

Me once more...I am trying to set up a bCNC Linux install with your very intricate source for handling the PRUs on the BeagleBone Black. 

`sudo apt install bcnc` installs the source on the board. I see on the board and via bcnc, there is no way to communicate their whereabouts well enough to pick and choose address:port so far.

Is this something that was taken out of beagleg? I mean to say, was the beagleg source made with it or without the server alongside the GUI?

Seth

P.S. I will keep attempting to make the debian install of bcnc work with the beagleg source. I guess working on a Linux box will help more than not too. I am trying to tie up some loose ends and get data on what route I need to take next.

silver2row, 2025-01-30:
I will be leaving soon. I will return later to find a reply (hopefully). I know what you told me last time about `do not hold your breathe`. So, I will be patient.

closed 2025-01-30 by silver2row

silver2row, 2025-01-30:
I think it still works! It works!

hzeller, 2025-01-30:
I didn't know about bCNC until you brought it up; it looks like a neat frontend!

It seems to be focused on sending gcode to whatever backend and not necessarily concerned about the PRU.
It mentions GRBL as a potential backend and that is typically using a serial interface. If you connect to BeagleG, that will be a TCP port. So if bCNC can talk to a TCP port, you can control a BeagleG-driven machine with bCNC.

silver2row, 2025-01-31:
Right...

I have been able to connect. The issue now is setting up the mandatory end stops and main electrical stop.

I have been able to use sockets in their bCNC build for communicating to the BBB with beagleg source.

Seth

P.S. I ordered some end stops/switches to handle the machine build in quest now. I can test with a smaller machine, i.e. that is already put together and ready to go. I will update you on the build once it is done and over. I am still reviewing the source to see if I need to alter or change specifics to get it up and running.

silver2row, 2025-02-05:
```
// ERROR: in soft-estop. Home and/or M999 first
X140.42Y17.92
ok
// ERROR: in soft-estop. Home and/or M999 first
X138.42Y16.27
ok
// ERROR: in soft-estop. Home and/or M999 first
X136.57Y14.82
ok
// ERROR: in soft-estop. Home and/or M999 first
X133.62Y12.68
ok
// ERROR: in soft-estop. Home and/or M999 first
X131.62Y11.32
ok
// ERROR: in soft-estop. Home and/or M999 first
X128.42Y9.32
ok
// ERROR: in soft-estop. Home and/or M999 first
X126.42Y8.18
ok
// ERROR: in soft-estop. Home and/or M999 first
X123.22Y6.47
ok
// ERROR: in soft-estop. Home and/or M999 first
X119.22Y4.57
ok
// ERROR: in soft-estop. Home and/or M999 first
X116.86Y3.58
ok
// ERROR: in soft-estop. Home and/or M999 first
X113.22Y2.17
ok
// ERROR: in soft-estop. Home and/or M999 first
X111.62Y1.63
ok
// ERROR: in soft-estop. Home and/or M999 first
X107.22Y0.28
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X104.65Y-0.38
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X101.62Y-1.07
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X98.42Y-1.67
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X96.82Y-1.92
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X93.22Y-2.37
ok
// ERROR outside machine limit: Axis Y < 0. Ignoring move!
X89.22Y-2.72
ok
// ERROR: in soft-estop. Home and/or M999 first
G0Z2.54
ok
S0
ok
M5
ok
// BeagleG: set_temperature(0.0) not implemented.
M30
ok
```

Are these errors due to the end stops or something else completely? I seem to think that my part is out of bounds for now but I am not sure how to use this front end well enough yet.

Those are the errors from the building of a part so far.

silver2row, 2025-02-05:
Update...I figured out the soft-estop error. Now, it is about homing. Blah...

Is there a file that saves all the logs from beagleg?

silver2row, 2025-02-07:
I find that this error for G28 is my concern...

```
// Line 3: G-Code Syntax Error: Letter '?' is not followed by a number but '????????????????????????????????????????????G28'
G28
ok
```

is that from the compilation or is there a known bug in the source?